donovan6000/M33-Fio #196: Connection Problem Raspberry PI3 and Micro3D

andershem, 2016-10-30:
Hi

I can't get the m3d printer to connect, i have installed Octoprint on a raspberry pi 3 and installed the M33-Fio plugin. I followed this guide (Basic setup) to install Octoprint. (https://github.com/foosel/OctoPrint/wiki/Setup-on-a-Raspberry-Pi-running-Raspbian)

But when i try to connect to the printer I get strange behaviors.
It first connects to /dev/ttyACM0 interface, then the interface disappear and reappear soon after as /dev/ttyACM1. Then the process repeats itself (this time, ttyACM1 disappears and ttyACM0 reappears).

I have tried to disconnect the printer, change usb ports, reboots of printer and raspberry pi. but i don't get the printer to connect.

I'm using:
Octoprint Version: 1.2.16 (master branch)
M33 Fio (1.10)
Rasbian (Kernel 4.4.26-v7+)
M3D Firmware Version: 2016-04-04-02

Serial log from Octoprint:

2016-10-30 20:01:36,537 - SERIAL - DEBUG - Changing monitoring state from 'Offline' to 'Detecting serial port'
2016-10-30 20:01:36,566 - SERIAL - DEBUG - Connecting to: /dev/ttyACM0
2016-10-30 20:01:36,568 - SERIAL - DEBUG - Changing monitoring state from 'Detecting serial port' to 'Opening serial port'
2016-10-30 20:01:36,569 - SERIAL - DEBUG - Connected to: Serial<id=0x6f2133d0, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=None, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
2016-10-30 20:01:36,570 - SERIAL - DEBUG - Starting baud rate detection
2016-10-30 20:01:36,571 - SERIAL - DEBUG - Changing monitoring state from 'Opening serial port' to 'Detecting baudrate'
2016-10-30 20:01:36,586 - SERIAL - DEBUG - Trying baudrate: 115200
2016-10-30 20:01:36,612 - SERIAL - DEBUG - Changing monitoring state from 'Detecting baudrate' to 'Offline'
2016-10-30 20:01:36,658 - SERIAL - DEBUG - Connection closed, closing down monitor
2016-10-30 20:01:38,296 - SERIAL - DEBUG - Connecting to: /dev/ttyACM1
2016-10-30 20:01:43,309 - SERIAL - DEBUG - Connecting to: /dev/ttyACM1
2016-10-30 20:01:43,314 - SERIAL - DEBUG - Changing monitoring state from 'Offline' to 'Error: Connection error, see Terminal tab'
2016-10-30 20:01:43,316 - SERIAL - DEBUG - Unexpected error while connecting to serial port: /dev/ttyACM1 OSError: '[Errno 16] Device or resource busy: '/dev/ttyACM1'' @ comm.py:_openSerial:1545 (hook default)
2016-10-30 20:01:43,319 - SERIAL - DEBUG - Send: N0 M110 N0*125
2016-10-30 20:07:59,682 - SERIAL - DEBUG - Changing monitoring state from 'Offline' to 'Detecting serial port'
2016-10-30 20:07:59,704 - SERIAL - DEBUG - Connecting to: /dev/ttyACM1
2016-10-30 20:07:59,706 - SERIAL - DEBUG - Changing monitoring state from 'Detecting serial port' to 'Opening serial port'
2016-10-30 20:07:59,708 - SERIAL - DEBUG - Connected to: Serial<id=0x6f25ab10, open=True>(port='/dev/ttyACM1', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=None, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
2016-10-30 20:07:59,709 - SERIAL - DEBUG - Starting baud rate detection
2016-10-30 20:07:59,709 - SERIAL - DEBUG - Changing monitoring state from 'Opening serial port' to 'Detecting baudrate'
2016-10-30 20:07:59,723 - SERIAL - DEBUG - Trying baudrate: 115200
2016-10-30 20:07:59,731 - SERIAL - DEBUG - Recv: e1
2016-10-30 20:07:59,772 - SERIAL - DEBUG - Changing monitoring state from 'Detecting baudrate' to 'Offline'
2016-10-30 20:07:59,817 - SERIAL - DEBUG - Connection closed, closing down monitor
2016-10-30 20:08:06,169 - SERIAL - DEBUG - Send: N0 M110 N0*125

donovan6000, 2016-10-30:
The jumping between different /dev/tty ports is normal since M33 Fio has to switch the printer into firmware mode which causes the host to reenumerate the printer, and that usually assigns the printer to a different /dev/tty port..

Make sure you have the [udev rule for the Micro 3D printer](https://github.com/donovan6000/M33-Fio/blob/master/installers/Linux/90-micro-3d-local.rules) installed. It sounds like some other program, probably ModemManager, is trying to communicate with the printer whenever it's enumerated, and the udev rule prevents that from happening.

andershem, 2016-10-30:
Thank you, the udev rule solved the problem.
Printer is working and printing now.

closed 2016-10-30 by andershem