donovan6000/M33-Fio #198: version update

analogmonster, 2016-11-06:
was on 1.7.  Octoprint was reporting m33-fio as up to date in software update tab, even after clicking check for update now button.  Updated to 1.11 master manually 5 mins ago and it works fine.  Any idea why my setup may have failed to detect updates (it seems since early September its not seen your updates).

On a side note it appears to have fixed an intermittent start up fault I had where the terminal would report unknown gcode - I believe it was related to the heat bed.  Just had to click connect again and it would work fine the second time, but now seems to have gone totally.

donovan6000, 2016-11-06:
I have no idea why it wouldn't detect that a newer version is available. M33 Fio uses the [plugin update hook described in OctoPrint's documentation](https://github.com/foosel/OctoPrint/wiki/Plugin:-Software-Update#hooks), so there shouldn't be any issues with detecting new versions.

I tried recreating your issue by installing M33 Fio V1.7, and OctoPrint was successfully able to detect that M33 Fio version 1.11 was available. So OctoPrint probably wasn't updating your cache for some reason.
[image]

analogmonster, 2016-11-06:
Ok I will close this here since it seems to be an octoprint rather than m33-fio related thanks

closed 2016-11-06 by analogmonster