donovan6000/M33-Fio #199: support option

analogmonster, 2016-11-09:
I customise my profiles in cura 15.04.  If I have it set to no support and I open an stl in octoprint for slicing, support and model on model support are unticked, but the final gcode does contain the support.  To actually print without supports I have to select then deselect support.  It would be convient for me if m3-fio remembered I dont usually use supports.  Thanks

donovan6000, 2016-11-09:
I tried to recreate this issue, but I haven't been able to get it to generate supports when the support checkboxes aren't ticked.

[Here's](https://github.com/donovan6000/M33-Fio/files/582079/micro_3d_pla_no_support.zip) the Cura profile I used when trying to recreate this. All I did to create the profile was click the 'Save' button at the bottom left of M33 Fio's profile editor to save a local copy of the slicer profile being used. Then I imported that profile into Cura and changed the `Support type` setting from `Everywhere` to `None`. Then I exported the profile from Cura and imported it into OctoPrint with the Settings > CuraEngine (<= 15.04) > "Import Profile..." button. Then I sliced a file using that new profile and didn't change any slicer settings in M33 Fio's profile editor while doing so.

The resulting command run by OctoPrint was this:
```
/usr/share/cura/CuraEngine -v -p -s initialSpeedupLayers=4 -s minimalFeedrate=12 -s preSwitchExtruderCode=;Switch between the current extruder and the next extruder, when printing with multiple extruders.\n;This code is added before the T(n)\n -s retractionAmountExtruderSwitch=14500 -s supportXYDistance=700 -s insetXSpeed=14 -s retractionZHop=100 -s raftSurfaceLinewidth=350 -s postSwitchExtruderCode=;Switch between the current extruder and the next extruder, when printing with multiple extruders.\n;This code is added after the T(n)\n -s retractionSpeed=20 -s raftAirGapLayer0=350 -s raftLineSpacing=2000 -s filamentFlow=100 -s raftBaseLinewidth=2500 -s infillOverlap=15 -s inset0Speed=12 -s coolHeadLift=1 -s extrusionWidth=350.0 -s upSkinCount=9 -s initialLayerSpeed=8 -s raftSurfaceLayers=2 -s raftInterfaceLinewidth=500 -s raftInterfaceThickness=200 -s minimalLayerTime=0 -s infillSpeed=16 -s supportExtruder=-1 -s fanSpeedMax=255 -s supportType=1 -s enableCombing=1 -s fanSpeedMin=255 -s supportZDistance=150 -s supportEverywhere=0 -s filamentDiameter=1750 -s initialLayerThickness=300 -s supportAngle=-1 -s fanFullOnLayerNr=1 -s raftBaseThickness=400 -s raftMargin=2000 -s layerThickness=250 -s endCode=\n -s minimalExtrusionBeforeRetraction=100 -s raftSurfaceLineSpacing=315 -s retractionMinimalDistance=1500 -s objectSink=0 -s retractionAmount=2400 -s skirtLineCount=0 -s startCode=M109 T0 S215.0\nT0\n; Sliced at: Wed 09-11-2016 14:46:10\n; Basic settings: Layer height: 0.25 Walls: 1.4 Fill: 8.75\n -s raftInterfaceLineSpacing=1000 -s skirtDistance=0 -s raftSurfaceSpeed=8 -s printSpeed=16 -s fixHorrible=0 -s layer0extrusionWidth=350 -s moveSpeed=20 -s raftBaseSpeed=8 -s supportLineDistance=1750 -s posx=52000 -s posy=55000 -s sparseInfillLineDistance=4000 -s insetCount=4 -s downSkinCount=9 -s raftFanSpeed=100 -s multiVolumeOverlap=0 -s raftSurfaceThickness=200 -o /tmp/tmpwT512j.gco /home/donovan6000/.octoprint/uploads/digivice.stl
```
I examined the resulting G-code in Cura and verified that it didn't have any supports.
[image]

Then I compared how this new profile differs from the original by slicing using the original profile that the new one was based off of. I did the same thing where I didn't change any slicer settings in M33 Fio's profile editor while slicing with the original profile.

The resulting command run by OctoPrint was this:
```
/usr/share/cura/CuraEngine -v -p -s initialSpeedupLayers=4 -s minimalFeedrate=12 -s preSwitchExtruderCode=;Switch between the current extruder and the next extruder, when printing with multiple extruders.\n;This code is added before the T(n)\n -s retractionAmountExtruderSwitch=14500 -s supportXYDistance=700 -s insetXSpeed=14 -s retractionZHop=100 -s raftSurfaceLinewidth=350 -s postSwitchExtruderCode=;Switch between the current extruder and the next extruder, when printing with multiple extruders.\n;This code is added after the T(n)\n -s retractionSpeed=20 -s raftAirGapLayer0=350 -s raftLineSpacing=2000 -s filamentFlow=100 -s raftBaseLinewidth=2500 -s infillOverlap=15 -s inset0Speed=12 -s coolHeadLift=1 -s extrusionWidth=350.0 -s upSkinCount=9 -s initialLayerSpeed=8 -s raftSurfaceLayers=2 -s raftInterfaceLinewidth=500 -s raftInterfaceThickness=200 -s minimalLayerTime=0 -s infillSpeed=16 -s supportExtruder=-1 -s fanSpeedMax=255 -s supportType=1 -s enableCombing=1 -s fanSpeedMin=255 -s supportZDistance=150 -s supportEverywhere=1 -s filamentDiameter=1750 -s initialLayerThickness=300 -s supportAngle=50 -s fanFullOnLayerNr=1 -s raftBaseThickness=400 -s raftMargin=2000 -s layerThickness=250 -s endCode=\n -s minimalExtrusionBeforeRetraction=100 -s raftSurfaceLineSpacing=315 -s retractionMinimalDistance=1500 -s objectSink=0 -s retractionAmount=2400 -s skirtLineCount=0 -s startCode=M109 T0 S215.0\nT0\n; Sliced at: Wed 09-11-2016 14:47:09\n; Basic settings: Layer height: 0.25 Walls: 1.4 Fill: 8.75\n -s raftInterfaceLineSpacing=1000 -s skirtDistance=0 -s raftSurfaceSpeed=8 -s printSpeed=16 -s fixHorrible=0 -s layer0extrusionWidth=350 -s moveSpeed=20 -s raftBaseSpeed=8 -s supportLineDistance=1750 -s posx=52000 -s posy=55000 -s sparseInfillLineDistance=4000 -s insetCount=4 -s downSkinCount=9 -s raftFanSpeed=100 -s multiVolumeOverlap=0 -s raftSurfaceThickness=200 -o /tmp/tmpTuAugf.gco /home/donovan6000/.octoprint/uploads/digivice.stl
```

I examined the resulting G-code in Cura and verified that it did have supports.
[image]

I compared the commands that were run when using the different profiles and saw that the `supportEverywhere` and `supportAngle` Cura Engine parameters differed. That's to be expected since [those are the parameters that OctoPrint's Cura plugin uses the `support` Cura profile setting to set](https://github.com/foosel/OctoPrint/blob/master/src/octoprint/plugins/cura/profile.py#L866-L867). So it looks like everything is working correctly. Can you upload the Cura profile that your using when you experience this issue?

analogmonster, 2016-11-10:
Thanks for the thorough reply.  I apologise it seems to be an issue with my PI or SD card, some plugins aren't loading properly and a profile keeps coming back after I delete it.  Maybe something went wrong with the recent update.  I will start with a fresh file system.  Thanks again

closed 2016-11-10 by analogmonster

donovan6000, 2016-11-10:
M33 Fio will reinstall all its provided slicer profiles every time OctoPrint starts up and whenever a client connects to OctoPrint, so trying to delete them or overwrite them with different profiles that have the same name will only temporarily work until they are reinstalled by M33 Fio. So make sure that any new profiles that you import don't share the same name as M33 Fio's profiles.