donovan6000/M33-Fio #200: Slicing Prints

bobbypop999, 2016-11-19:
When you have got to the point when you choose the quality and infill the manual settings it only shows infill, layerheight. I am using the currant version of octoprint and M33 Fio, safari mobile.
If I use Crome it has more options like the version before.

donovan6000, 2016-11-21:
Can you post a screenshot? I don't have access to anything that can run safari mobile, so I might not be able to fix this myself unless it's something simple like adding vendor prefix names to some of the CSS properties.

dieideeistgut, 2016-11-23:
I am running into several issues with Safari and also Firefox too - current Chrome behaves nice though. I'm actually thinking that these issues come from within octoprint and not from M33-Fio.

donovan6000, 2016-11-24:
Can you guys test out M33 Fio's devel version and let me know if any of the visual issue were resolved?

You can install it by install a plugin from this url `https://github.com/donovan6000/M33-Fio/archive/devel.zip` using OctoPrint's builtin plugin manager.

bobbypop999, 2016-11-24:
Yes I will have a go but it will be a wekk before I get back to you.

bobbypop999, 2016-12-02:
Unfortuanlly it doesn't work but here is a screen shot for more info.
[image]

bobbypop999, 2016-12-02:
Found another problem! After the first page of slicing you can't even get to the quality settings it just starts slicing?

donovan6000, 2016-12-03:
I'm not going to be able to fix this myself since I don't have access to anything running an older version of Safari mobile. M33 Fio just wasn't designed for outdated web browsers, and the version of Safari mobile that your using doesn't support some of the modern CSS features that M33 Fio uses.

If anyone's interested in trying to fix this issue, then [here's M33 Fio's css file](https://github.com/donovan6000/M33-Fio/blob/master/octoprint_m33fio/static/css/m33fio.css).

bobbypop999, 2016-12-03:
Ok my iPad 2 is outdated so if I could upgrade to ios10 maybe that could solve issues?

donovan6000, 2016-12-03:
If ios10 comes with a newer version of Safari mobile then it could fix the issue.

bobbypop999, 2016-12-03:
Just realised I have access to iOS 10 on another doe use I will try it immediately.

bobbypop999, 2016-12-03:
Tried but didn't work!!! Same screen as last time argh!!!

bobbypop999, 2016-12-03:
I have just tried the slicer plug in works very good also it can change the print speed and more.
Until this might get fixed I think I will use the plug in instead.

bobbypop999, 2016-12-03:
Now I only get this
[image]

donovan6000, 2016-12-04:
@bobbypop999 Update to the latest version [V1.13](https://github.com/donovan6000/M33-Fio/releases/tag/V1.13).

bobbypop999, 2016-12-04:
How I can't get to the settings?

donovan6000, 2016-12-04:
Your web browser is detecting a JavaScript error when parsing M33 Fio's JavaScript, and it halts JavaScript execution on the page when that happens. So you'll need to manually uninstall M33 Fio since you wont be able to do it from the web interface due to the JavaScript error.

If your using OctoPi, then you can SSH into the Raspberry Pi and run the following command to uninstall M33 FIo and restart OctoPrint. Then you'll be able to use the web interface again to install the latest version of M33 Fio.
```
/home/pi/oprint/bin/pip uninstall OctoPrint-M33Fio
service octoprint start
```

bobbypop999, 2016-12-04:
How do I ssh in Windows?

donovan6000, 2016-12-04:
You can use [Putty](https://www.raspberrypi.org/documentation/remote-access/ssh/windows.md).

bobbypop999, 2017-03-11:
I was just woundering if you were to have the heated bed that chase has made pluged into the raspberry pi after you slice a model and click print normaly a window that asks for nozzle temp and wave bonding comes up but with the heated bed does it also have bed temp there? Also could you add vase mode to the slicing?

bobbypop999, 2017-09-24:
Just to let you know on ios 11 every thing works fine!

[image]