donovan6000/M33-Fio #204: connection issue with Octoprint 1.3.0

JulienMorand, 2016-12-08:
I've made the update to octoprint 1.3.0 on a raspberry Pi. I can't connect to the printer anymore. I've look at the terminal and it seems that sent message is N0 M110 N0 for connection.

I've seen in a previous issue that should be only M110.

Do anyone have this issue?

bobbypop999, 2016-12-08:
I'm not getting that problem but when I start a print and the model is in the center it prints in the cornder or somewere else? It has just happed after I updated.

th0ma5w, 2016-12-09:
I am seeing this issue... keeps trying to detect the baud rate, sends the N0 M110 N0*125 ... I may try fiddling with reinstalling or something.

th0ma5w, 2016-12-09:
Update! Sorry for the quick double post, but removing the M33-Fio plugin and then reinstalling (all of this using the venv setup instructions, so it was just working with the Octoprint plugin manager) seemed to fix this for me. Probably going to be difficult to reproduce this bug, but reinstallation worked fine for me.

donovan6000, 2016-12-09:
For some reason M33 Fio's pre-processor stages aren't being applied when using OctoPrint V1.3.0. I'm still looking into why this is happening. This problem only affects people who are using a Micro 3D printer, so people using Micro 3D printers should stay at OctoPrint V1.2.18 until this gets resolved.

@JulienMorand I'm not experiencing any connection issues, so it's probably resulting from you using an older version of M33 Fio. Make sure you have the latest version of M33 Fio installed, which currently is [V1.14](https://github.com/donovan6000/M33-Fio/releases).

bobbypop999, 2016-12-09:
Is that why my printer doesnt home it self as a click print and now it stays where it was heats up and starts a print somewere random?

donovan6000, 2016-12-09:
@bobbypop999 Yes. The Preparation pre-processor stage is responsible for homing and priming the extruder, so what your experiencing is a result of the pre-processor stages not being applied.

bobbypop999, 2016-12-09:
Ok now I know my printer isnt broken.few😀

gromeo, 2016-12-09:
Hi,

as I already wrote on the Google plus M3D community **([this post here](https://plus.google.com/115441318241899642555/posts/99tPM6B7Qkk?sfc=true))**, I'm experiencing this problem too.
Donovan, do you know of an easy way to rollback to a previous version of Octoprint (I installed the OctoPi distribution)? The [official FAQ of octoprint](https://github.com/foosel/OctoPrint/wiki/FAQ#how-can-i-roll-back-to-an-earlier-version-after-an-update) says that I should use command line (I imagine doing so through ssh) and it refers to a "cd /path/to/OctoPrint" command: do you know what is this "/path/to/OctoPrint" in an octopi distro?

Many thanks,
Giovanni Romeo

donovan6000, 2016-12-09:
@gromeo I've never used OctoPi, so I'm not really sure where it keeps the checkout folder, but you can see where it is by going to OctoPrint's Settings > Software Updater > Wrench in top right > OctoPrint checkout folder.
[image]

gromeo, 2016-12-09:
Many thanks @donovan6000, I'll try to rollback this evening when I'll be back home from work.
By the way, did you find what changed with this new version of Octoprint that messed up m33?

Giovanni

donovan6000, 2016-12-09:
Not yet, but I'm free all day tomorrow so it shouldn't be more than a day or two before I find out what changed and update M33 Fio accordingly.

CatSailer1279, 2016-12-09:
Just to let you know: Same issue with my printer. With 1.3.0 Octoprint can not connect to printer. Stays in Connecting... I reinstalled M33 Fio. After that Octorpint can connect to my printer, but there is no homing before the print.
Many thanks in advance for your work! :)

autious, 2016-12-09:
I'm experience the issue similar to what donovan6000 is describing. My printer skips the initial printing and just stands still pushing out plastic on the spot. Can't find a simple way to downgrade octoprint.

th0ma5w, 2016-12-09:
I used the venv Octopi install instructions so at a high level I did the following:

1. Move ~/.octoprint to ~/.octoprint-bak and then created an empty .octoprint dir

2. Removed the Octoprint from venv/lib/python2.7/site packages/

3. From the git clone I did:
git checkout 1d0657a8ebf5ed7b22b2a3663ba13389dd500f6d

4. Then I did venv/bin/python setup.py install

Then I just reconfigured everything using the config.yaml from the backup as a reference, and reinstalled M33-Fio ...

I think what will work for you will depend on how you set up everything, but the above sha1 hash is a good working version. If you've cloned the repository that command will finish very quickly as the history already exists locally.

JulienMorand, 2016-12-09:
Some updates:

I've uninstalled then re-installed M33-Fio using the plugin manager on Octopi. I can connect to the printer now... But I am experiencing the same behavior as described above: standing  on a given location and pushing out plastic...

jaaron, 2016-12-10:
For reference, I experienced this problem as well.

To roll back on my Octopi installation:
```
pi@octopi:~ $ cd OctoPrint
pi@octopi:~/OctoPrint $ git checkout 1.2.18
pi@octopi:~/OctoPrint $ source ~/oprint/bin/activate
(oprint)pi@octopi:~/OctoPrint $ python setup.py clean
(oprint)pi@octopi:~/OctoPrint $ python setup.py install
(oprint)pi@octopi:~/OctoPrint $ ps ax | grep octoprint
 1389 ?        Sl     0:28 /home/pi/oprint/bin/python /home/pi/oprint/bin/octoprint --host=127.0.0.1 --port=5000
 1496 pts/0    S+     0:00 grep --color=auto octoprint
(oprint)pi@octopi:~/OctoPrint $ kill 1389
(oprint)pi@octopi:~/OctoPrint $ octoprint --daemon start -b ~/.octoprint/
```

jaaron, 2016-12-10:
And here's the stacktrace trying to load M33 Fio in Octoprint 1.3

> 2016-12-09 23:20:04,001 - octoprint.plugin - ERROR - Error while calling plugin m33fio
> Traceback (most recent call last):
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.0-py2.7.egg/octoprint/plugin/__init__.py", line 229, in call_plugin
>    result = getattr(plugin, method)(*args, **kwargs)
>  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_m33fio/__init__.py", line 5357, in on_event
>    self.setFileLocations()
>  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_m33fio/__init__.py", line 5238, in setFileLocations
>    self.savePrinterProfile()
>  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_m33fio/__init__.py", line 886, in savePrinterProfile
>    self._printer_profile_manager.save(printerProfile, True, not self._settings.get_boolean(["NotUsingAMicro3DPrinter"]))
>  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.0-py2.7.egg/octoprint/printer/profile.py", line 255, in save
>    self._save_to_path(self._get_profile_path(identifier), profile, allow_overwrite=allow_overwrite)
>  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.0-py2.7.egg/octoprint/printer/profile.py", line 370, in _save_to_path
>    raise InvalidProfileError()
> InvalidProfileError

donovan6000, 2016-12-10:
@jaaron What version of M33 Fio are you using? That InvalidProfileError error should be fixed in M33 FIo >= V1.12.

jaaron, 2016-12-10:
At the moment, 1.14, but I've done an uninstall reinstall since that exception was generated. OctoPrint reported it as being up to date when the exception was generated, but I've found that plugin updater a little iffy at times so I could easily have been on an old version.

CatSailer1279, 2016-12-11:
Just installed the version 1.15 and now my printer is working fine again. Thanks! :)

donovan6000, 2016-12-12:
All the OctoPrint V1.3.0 issues should now be fixed in M33 Fio [V1.15](https://github.com/donovan6000/M33-Fio/releases/tag/V1.15) if you guys want to try it.

gromeo, 2016-12-12:
I'll give it a try asap. 
Thanks!

closed 2016-12-20 by donovan6000