donovan6000/M33-Fio #205: Backlash Calibration Dialog Box Doesn't Work

fooforever, 2016-12-10:
I've just installed octoprint using their install for windows instructions on the wiki and installed m33-fio after that. (Did it this was because the install script didn't work for me for some reason.) I think I had a backlash issue on the last print I did so I wanted to do a test and I know m33-fio did this hence the reason for the octoprint install. however when I click the Run Backlash Calibration button this is what I get:
[image]
None of the buttons do anything and I can't exit the dialog because the cancel button isn't a button. This looks like a problem with the plugin to me and I can't find anything on the forum about this sort of issue.

dpsfotocestou, 2016-12-10:
Same problem after update to new version

donovan6000, 2016-12-10:
We're still working on [making M33 Fio compatible with OctoPrint V1.3.0](https://github.com/donovan6000/M33-Fio/issues/204#issuecomment-266207140). In the mean time you can downgrade to OctoPrint V1.2.18 or wait a few days until we get everything worked out.

donovan6000, 2016-12-12:
This should fixed now in M33 Fio [V1.15](https://github.com/donovan6000/M33-Fio/releases/tag/V1.15) if you want to try it.

closed 2016-12-13 by fooforever