donovan6000/M33-Fio #209: Cloning models messes up placement

supermaik, 2017-01-03:
I'm trying to print 6 of the same model on my M3D Micro.  When I go to position/clone the model in the slicer view it lets me move things around and position them within the bounding area and everything appears fine.  Once I load the gcode and try to print it, I get an error stating the model exceeds the print area.  When I look at the gcode viewer the parts are clearly outside the grid but I can't figure out why.  Even if I try to account for the placement by offsetting it in the slicer editor thing, it still puts them in the same spot.  Any idea what's causing this?

supermaik, 2017-01-03:
Ok I'll take part of that back.  I was able to trial and error the offset to get it to put the model within the grid.  I just hit start on the print job and looks like it's gonna go.  Just frustrating trying to get it to go where I want.

donovan6000, 2017-01-03:
This is fixed in [M33 Fio's devel branch](https://github.com/donovan6000/M33-Fio/tree/devel) now. You can install it by installing a plugin from the following URL using OctoPrint's builting plugins manager.
```
https://github.com/donovan6000/M33-Fio/archive/devel.zip
```

closed 2017-01-09 by donovan6000