donovan6000/M33-Fio #211: Keeps corrupting printer firmware

p0lar-bear, 2017-02-10:
I've compiled OctoPrint 1.3.1 from source on an old laptop (IBM Thinkpad T41) running Lubuntu 14.04.5 and have M33 Fio 1.17 installed through OctoPrint's plugin manager. My Micro 3D is one of the limited edition clear ones.

Every time I attempt to use OctoPrint with my printer, the firmware just seems to get corrupted. The printer will work just fine with the official M3D software on my Windows computer. I plug it into the Linux laptop, choose the port (usually /dev/ttyACM0), and when I click connect one of two things will happen:

1. This output/error in the console:
```
    Connecting to: /dev/ttyACM0
    Connecting to: /dev/ttyACM0
    Changing monitoring state from 'Offline' to 'Error: Connection error, see Terminal tab'
    Unexpected error while connecting to serial port: /dev/ttyACM0 OSError: '[Errno 16] Device or resource busy: '/dev/ttyACM0'' @ comm.py:_openSerial:1694 (hook default)
```

2. Baud rate detection fails:
```
    Connecting to: /dev/ttyACM1
    Changing monitoring state from 'Offline' to 'Opening serial port'
    Connected to: Serial<id=0xafd1254c, open=True>(port='/dev/ttyACM1', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=None, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
    Starting baud rate detection
    Changing monitoring state from 'Opening serial port' to 'Detecting baudrate'
    Trying baudrate: 115200
    Send: N0 M110 N0*125
    Recv: ??
    Baudrate test retry: 4
    Send: N0 M110 N0*125
    Recv: ??
    Baudrate test retry: 3
    Send: N0 M110 N0*125
    Recv: ??
    Baudrate test retry: 2
    Send: N0 M110 N0*125
    Recv: ??
    Baudrate test retry: 1
    Send: N0 M110 N0*125
    Recv: ??
    Baudrate test retry: 0
    Send: N0 M110 N0*125
    Recv: ??
    Changing monitoring state from 'Detecting baudrate' to 'Offline'
    Changing monitoring state from 'Offline' to 'Error: No more baudrates to test, and no suitable baudrate found.'
    Connection closed, closing down monitor
```

Either outcome results in corrupt firmware - both OctoPrint (if I click Connect again), and the official M3D software (if I plug it into my Windows computer afterwards) both prompt me to update the firmware.

donovan6000, 2017-02-10:
It looks like some other program, probably ModemManager, is trying to communicate with the printer at the same time as OctoPrint. Make sure you have the [Micro 3D's udev rule installed](https://github.com/donovan6000/M33-Fio/blob/master/installers/Linux/90-micro-3d-local.rules) and let me know if you still have any issues after that.

p0lar-bear, 2017-02-10:
That fixed the printer firmware corrupting.

But now when I click connect, the printer LED flashes twice and I still get the dialog saying that it couldn't connect. The status just says "Offline" and there's no error.

If I choose the port on the web interface and click connect, i get the same thing, but the port dropdown resets to "AUTO." If I have a terminal open on my laptop while this happens, it outputs `usb 3-2: string descriptor 0 read error: -32`, which suggests to me that the printer disconnects when the port is opened and OctoPrint attempts to communicate with it.

    Changing monitoring state from 'Offline' to 'Detecting serial port'
    Connecting to: /dev/ttyACM0
    Changing monitoring state from 'Detecting serial port' to 'Opening serial port'
    Connected to: Serial<id=0xaff37d8c, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=None, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
    Starting baud rate detection
    Changing monitoring state from 'Opening serial port' to 'Detecting baudrate'
    Recv: wait
    Recv: wait
    Recv: wait
    Recv: wait
    Recv: wait
    Trying baudrate: 115200
    Recv: e1
    Changing monitoring state from 'Detecting baudrate' to 'Offline'
    Connection closed, closing down monitor
    Send: N0 M110 N0*125

p0lar-bear, 2017-02-10:
Never mind... plugged the printer into a different USB port and it started working. Seems to prefer the ports on the laptop's dock over the ports on the side of it.

closed 2017-02-10 by p0lar-bear