donovan6000/M33-Fio #212: Printer grinding against solidified plastic when printing rift

abbradar, 2017-02-24:
Hi,

I'm not sure if this is me giving wrong settings to the printer, me doing a bad job at calibration or a software bug. When I print a model with rift printer severely grinds against the first thick printed layer (head visibly vibrates because of this). Later it stabilizes but not without a quality impact (first layers are squashed a bit). Do you have an idea what could go wrong? I did a manual calibration of Z0s, and test border looked fine. I leave a little more space between head and bed than you describe when calibrating (paper is just a little difficult to move, not impossible). My firmware is iMe V00.00.01.23.

abbradar, 2017-02-25:
Similar user report: https://www.thingiverse.com/groups/m3d-the-micro-3d-printer/topic:6479#comment-1001113

abbradar, 2017-02-25:
With stock firmware V2016040401 it doesn't grind, head still moves but very slightly and rarely. I think that's  a possible bug in iMe but I'm not sure -- can you confirm? Check on [this model](https://abbradar.net/me/share/scaled_part.stl), with default settings (medium quality/fill) and a raft.

donovan6000, 2017-04-26:
It sounds like your Z0 position has been calibrated slightly to low. You can manually set the Z0 position by positioning the nozzle at a desired height above the center of the bed and clicking the 'Save Z as bed center Z0' button in the Calibration section of OctoPrint's Control tab.

donovan6000, 2017-06-02:
It's been a while since anyone has responded to this issue, so I'll probably close it in a couple of days. Let me know if you still have any interest in testing/fixing this issue.

closed 2017-06-05 by donovan6000