donovan6000/M33-Fio #214: How to host a camera

micdenny, 2017-03-14:
I saw the settings here:

[image]

but after I save and reboot octoprint I can't see anything on http://localhost:4999/snapshot.jpg

Have I missed a step?

donovan6000, 2017-04-26:
Hosting a camera should just work as soon as you enable that setting. It automatically configures OctoPrint's settings to use the URLs that it hosts, and you should be able to verify that it's all working by testing the URLs in OctoPrint's Settings > Webcam & Timelapse tab.

You can also try out the newest version, [V1.18](https://github.com/donovan6000/M33-Fio/releases/tag/V1.18), and let me know if your still having this issue. There was an issue with the camera hosting on macOS Sierra that was fixed.

micdenny, 2017-04-26:
I simply configured octopi setting the right property for my cam and now I can see the camera, but without using this M33-Fio feature, that I still don't understand what it adds, it simply helps you configure octoprint settings? I just did this: https://github.com/foosel/OctoPrint/wiki/Webcams-known-to-work

donovan6000, 2017-04-26:
M33 Fio is capable of hosting the webcam stream and snapshot itself without having to use any other programs, like mjpg-streamer.

Does running the following show any errors?
```
wget https://raw.githubusercontent.com/donovan6000/M33-Fio/master/octoprint_m33fio/webcam_server.py
python webcam_server.py
```

micdenny, 2017-04-26:
> M33 Fio is capable of hosting the webcam stream and snapshot itself without having to use any other programs, like mjpg-streamer.

thank you for the explanation, now it's clear.

> Does running the following show any errors?

I will test it and let you know.

donovan6000, 2017-06-02:
It's been a while since anyone has responded to this issue, so I'll probably close it in a couple of days. Let me know if you still have any interest in testing/fixing this issue.

micdenny, 2017-06-02:
Testing now, sorry...

micdenny, 2017-06-02:
@donovan6000 I don't see the above screenshot anymore with the new version, I only see this menu:

[image]

micdenny, 2017-06-02:
> Does running the following show any errors?

I get this error:

[image]

```
pi@octopi:~ $ python webcam_server.py
Traceback (most recent call last):
  File "webcam_server.py", line 19, in <module>
    from PIL import Image
ImportError: No module named PIL
```

micdenny, 2017-06-02:
I did this `sudo apt-get install python-imaging` and now the error is:

```
pi@octopi:~ $ python webcam_server.py
Traceback (most recent call last):
  File "webcam_server.py", line 20, in <module>
    import pygame.camera
ImportError: No module named pygame.camera
```

micdenny, 2017-06-02:
I did also `sudo apt-get install python-pygame` and now I get this, so it seems to work:

```
pi@octopi:~ $ python webcam_server.py
Using webcam device /dev/video0 with a resolution of 640x480 running at 20 frames/second
Hosting webcam still image at http://192.168.50.161:4999/snapshot.jpg
Hosting webcam video stream at http://192.168.50.161:4999/stream.mjpg
```

now I check the portal

micdenny, 2017-06-02:
but I can't see from the site where to find the menu to activate the cam anymore

donovan6000, 2017-06-03:
The menu is in M33 Fio's settings

micdenny, 2017-06-03:
I don't see it, what could be the reason? Do you check if there's a webcam to show the options?

donovan6000, 2017-06-05:
[M33 Fio checks if a camera is connected and all required libraries are installed](https://github.com/donovan6000/M33-Fio/blob/master/octoprint_m33fio/__init__.py#L6074-L6092) before showing the camera options. I wonder how you managed to see it before when you didn't had pygame and pillow installed though.

micdenny, 2017-06-05:
> M33 Fio checks if a camera is connected and all required libraries are installed before showing the camera options.

first question is why I had to install pygame manually?

> I wonder how you managed to see it before when you didn't had pygame and pillow installed though.

I wonder me too, probably something went mad after I upgraded to a new version of octopi, then I had to install a custom version of M33 from one of the maintainers of octopi to let the M33 plugin work again, then you released the compatible version and I installed your official distribution, and I think at the time I installed the custom M33 version or lately your official version the menu disappeared.

I'm quite confused right now, I think it is better to backup my octopi device, and start from scratch and see if everything works as expected or not.

donovan6000, 2017-06-09:
There were too many problems with automatically installing the requirements for M33 Fio's camera features and I didn't want people to have any problems installing M33 Fio, so that's why it's necessary to manually install the camera dependencies in order to use it.

Let me know if starting from scratch fixed the issue.

closed 2017-08-12 by donovan6000