donovan6000/M33-Fio #215: Have OctoPrint remember instances upon restart

DynamicMetaFlow, 2017-03-16:
Greetings, thank you for your contribution to our community. 

I use M33-Fio alongside OctoPrint in my classroom to control 5 printers. At times, the Raspberry Pi needs to be restarted and when this happens Octoprint's instances that M33-Fio provides are forgotten and needs to be initialized again. 

Would it be possible for M33-Fio to remember the instances that were created upon restart?

donovan6000, 2017-04-26:
The newest version of M33 Fio, [V1.18](https://github.com/donovan6000/M33-Fio/releases/tag/V1.18), allows saving an instance's settings and creating a new instance using those settings later. That should allow much more flexibility with maintaining instances whenever you have to restart the Raspberry Pi. Hope that helps.

donovan6000, 2017-06-02:
It's been a while since anyone has responded to this issue, so I'll probably close it in a couple of days. Let me know if you still have any interest in testing/fixing this issue.

closed 2017-06-05 by donovan6000