donovan6000/M33-Fio #216: Issue with OctoPrint version 1.3.2

loglow, 2017-03-17:
After updating OctoPrint to version 1.3.2, I can no longer connect to my M3D Micro printer. The terminal shows repeated "wait" commands being received, and the connection attempt never times out, it just hangs. Similar behavior can be observed when trying to connect to the M3D printer **without** the M33 Fio plugin enabled. After rolling back to OctoPrint version 1.3.1, everything works fine again. Please let me know if you'd like any additional info from me, and thanks very much.

BleedTheWay, 2017-03-17:
I too am having the same issue after updating to OctoPrint version 1.3.2. Only I am no longer receiving wait messages, I am simply getting "Recv: ??" after every connection attempt before OctoPrint gives up.

ashpool, 2017-03-18:
I updated to `OctoPrint 1.3.2` and `M33 Fio 1.17` and since then I'm unable to connect to the M3D printer, hangs on "Detecting baudrate":

[image]

The log:
```
2017-03-18 20:59:36,252 - octoprint.server - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://192.168.42.29/'}
2017-03-18 20:59:43,113 - octoprint.filemanager - INFO - Added 0 items from storage type "local" to analysis queue
2017-03-18 20:59:58,479 - octoprint.server - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://192.168.42.29:5001/'}
2017-03-18 21:00:03,188 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.42.124
2017-03-18 21:00:04,460 - octoprint.printer.profile - WARNING - Profile invalid, missing keys. Expected: ['name', 'color', 'axes', 'heatedBed', 'volume', 'model', 'id', 'extruder']. Actual: ['name', 'color', 'axes', 'heatedBed', 'volume', 'model', 'id', 'extruder']
2017-03-18 21:00:04,461 - octoprint.plugin - ERROR - Error while calling plugin m33fio
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/plugin/__init__.py", line 229, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_m33fio/__init__.py", line 5392, in on_event
    self.setFileLocations()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_m33fio/__init__.py", line 5273, in setFileLocations
    self.savePrinterProfile()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_m33fio/__init__.py", line 887, in savePrinterProfile
    self._printer_profile_manager.save(printerProfile, True, not self._settings.get_boolean(["NotUsingAMicro3DPrinter"]))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/printer/profile.py", line 259, in save
    self._save_to_path(self._get_profile_path(identifier), profile, allow_overwrite=allow_overwrite)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.2-py2.7.egg/octoprint/printer/profile.py", line 374, in _save_to_path
    raise InvalidProfileError()
InvalidProfileError
```

dwarfc, 2017-03-18:
I too have the same issue after updating to octoprint 1.3.2.

loglow, 2017-03-18:
In case you're having this problem, here are the (reasonably easy, tested and working) instructions to roll back to OctoPrint 1.3.1 in the meantime until this issue is fixed: 

https://github.com/foosel/OctoPrint/wiki/FAQ#how-can-i-revert-to-an-older-version-of-the-octoprint-installation-on-my-octopi-image

foosel, 2017-03-20:
The issue probably arises due to the plugin attempting to create/update a printer profile that thanks to not being merged over the default printer profile like it would be when creating it through the UI now lacks a parameter that was introduced with 1.3.2. The corresponding commit happened back in January. 

In light of this I'd seriously recommend people running with this plugin to please also test the release candidates I put out before every stable release. This issue could have *easily* been found and circumvented if anyone had done that and reported back. Avoiding problems like this is the whole point of going through the hassle of release candidates in the first place, and it requires help from the community.

With that being said: Without being able to test this due to the lack of an M3D printer, PR #217 should fix this problem I hope. If it does, installing the plugin from `https://github.com/foosel/M33-Fio/archive/fix/issue216.zip` (after uninstalling it first through OctoPrint's plugin manager) should also make it work against OctoPrint 1.3.2 until a new official release of this plugin is out (which it then should be able to update to without any manual intervention). No guarantees though, again, I couldn't actually test this.

ashpool, 2017-03-20:
> This issue could have easily been found and circumvented if anyone had done that and reported back

I'm happy to help if I can

foosel, 2017-03-20:
@ashpool awesome! Then please [read here](https://github.com/foosel/OctoPrint/wiki/Using-Release-Channels) on how to switch to the "Maintenance RCs" release channel, where Release Candidates for the next stable release are released on and you'll get an update notification for new RCs just like you get for stable releases :)

Also either follow [the OctoBlog](http://octoprint.org/blog/) or follow OctoPrint on [Twitter](https://facebook.com/OctoPrint), [G+](https://plus.google.com/+OctoprintOrg/posts) or [Facebook](https://facebook.com/OctoPrint) - I always post release announcements for release candidates there ([example](http://octoprint.org/blog/2017/03/10/new-release-candidate-1.3.2rc1/)), so you'll see when something new shows up. Btw, you can also follow the OctoBlog right from within OctoPrint - go to Settings > Announcements and enable at least "OctoPrint Release Announcements".

ashpool, 2017-03-20:
> installing the plugin from https://github.com/foosel/M33-Fio/archive/fix/issue216.zip

It works! 👏  🌟

Titofg, 2017-03-20:
Hello foosel and ashpool. Thanks for works and help !

Just a precision **for very newbie like me** on Octoprint : 
To install fix M33 plugin go to Octoprint plugin manager -> get more -> copy paste the link above from foosel and ashpool on URL box and click install

(Don't upload directly all files like me the first time... Oo)

foosel, 2017-03-21:
Ok, so steps to install my fixed version until the new official version is out (thanks @ashpool for testing):

1. Settings > Plugin Manager, uninstall your current M33Fio plugin
2. Open "Get more...", paste `https://github.com/foosel/M33-Fio/archive/fix/issue216.zip` into the "from URL" field, click "Install"

That should be all. If not already done, update OctoPrint to 1.3.2 (Settings > Software Update > Check for updates).

foosel, 2017-03-31:
Since there is no fixed release out yet, I've disabled the plugin on the repository [for now](https://github.com/OctoPrint/plugins.octoprint.org/commit/8753172daf9e72cdd811362695f8a27fe58e61ff), with a link back to this ticket, see [here](http://plugins.octoprint.org/plugins/m33fio/).

@donovan6000 I didn't want to simply redirect installations through the repo to my fixed version without any feedback about that from you, so I opted for the only currently available "something's wrong with this one" approach for repo listings (new options for handling situations like this one will be in 1.3.3). I can re-enable it and temporarily replace the archive URL with my fixed version though if you don't have the time to merge the available PR and release a new version right now. I just need to know.

theurgy, 2017-04-02:
@foosel: Thanks a lot for the fix! :1st_place_medal: It works well with the Micro 3D again. I am running OctoPrint 1.3.2 on a Raspberry Pi (generation 1 model B) (not OctoPi installation but an ordinary Raspbian 8.0 jessie). Uninstalled the M33Fio plugin as you described (21 mars 9:30 CET), and installed the fix all through the GUI of OctoPrint(restarted as instructed after uninstall and install). I've just printed a test piece, after a complete calibration without a hazzle. Thanks!

jbosse, 2017-04-02:
I just discovered OctoPi and this plugin this morning. I figured I was just a n00b. Installing from issue216 zip works for me. Trying my first print now :-)

dart27, 2017-04-09:
Can confirm the fix (https://github.com/foosel/M33-Fio/archive/fix/issue216.zip) and reinstall of m33fio works to resolve the issue. Wish I had Googled this up 2 weeks ago... Still awesome coding and much love!

closed 2017-04-26 by donovan6000

NiklasOlofsson, 2017-04-26:
Just want to say I love this plugin (I do not have the printer mentioned). Great stuff!

PR #217 (2017-04-25): Add missing printer profile prop & merge with default
This should fix #216. I was not able to test this though due to lack of a proper test environment (M3D).
commit bdf2422d (2017-04-25): Add missing printer profile prop & merge with default

This should fix #216. I was not able to test this though due to
lack of a proper test environment (M3D).