donovan6000/M33-Fio #218: Printer wont connect - No suitable baudrate found

noo-m3d, 2017-04-08:
I used to use my m3d often, but due to other reasons i have not used it for a month, i tried to use it recently and it gave me a error when i tried to connect.

Connecting to: /dev/ttyACM0
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0x6db9f150, open=True>(port='/dev/ttyACM0', baudrate=57600, bytesize=8, parity='N', stopbits=1, timeout=None, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: N0 M110 N0*125
Changing monitoring state from 'Connecting' to 'Offline'
Changing monitoring state from 'Offline' to 'Detecting serial port'
Connecting to: /dev/ttyACM0
Changing monitoring state from 'Detecting serial port' to 'Opening serial port'
Connected to: Serial<id=0x6db662f0, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=None, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Starting baud rate detection
Changing monitoring state from 'Opening serial port' to 'Detecting baudrate'
Trying baudrate: 115200
Send: N0 M110 N0*125
Connection closed, closing down monitor
Recv: ??
Baudrate test retry: 4
Send: N0 M110 N0*125
Recv: ??
Baudrate test retry: 3
Send: N0 M110 N0*125
Recv: ??
Baudrate test retry: 2
Send: N0 M110 N0*125
Recv: ??
Baudrate test retry: 1
Send: N0 M110 N0*125
Recv: ??
Baudrate test retry: 0
Send: N0 M110 N0*125
Recv: ??
Changing monitoring state from 'Detecting baudrate' to 'Offline'
Changing monitoring state from 'Offline' to 'Error: No more baudrates to test, and no suitable baudrate found.'
Connection closed, closing down monitor

showing 114 lines
Scroll to end | Select all

I am using an m3d connected to a octopi on a pi 3

noo-m3d, 2017-04-10:
I have solved it by using by reinstalling m33 fio from here https://github.com/foosel/M33-Fio/archive/fix/issue216.zip

donovan6000, 2017-04-26:
It should be working in the newest release, [V1.18](https://github.com/donovan6000/M33-Fio/releases/tag/V1.18).

closed 2017-04-26 by donovan6000

stwf, 2017-04-29:
I just upgraded to 1.18 and the problem is back. Gets stuck on Detecting baudrate forever. Using a micro3d on an RPI2.

ashpool, 2017-05-05:
I just updated to 1.18 and now I'm getting `Error: Failed to autodetect serial port, please set it manually.` in OctoPrint 1.3.2