donovan6000/M33-Fio #221: Slic3r Prusa Edition and saving default values.

NiklasOlofsson, 2017-04-26:
Hello,

I love this plugin. Great work. I use it with Slic3er Prusa Edition, I guess I loose some of the Prusa specific settings? The 2nd screen show some as unknown with my prusa edition ini:
[image]

Could this be corrected to support the Prisa Edition?

One other thing, could the first screen remember the last settings the user used?
I only have one slicer activated but I still need to select it from the drop down box.
[image]

donovan6000, 2017-04-27:
The quality and fill detection are kinda specific, and are based on the official software for the Micro 3D printer. So they were used in M33 Fio to make the transition easier for those using a Micro 3D printer. [Here's how the settings are detected for the Slic3r profiles](https://github.com/donovan6000/M33-Fio/blob/master/octoprint_m33fio/static/js/m33fio.js#L6692-L6834). So an extra low quality would only be detected if layerHeight is 0.35 and firstLayerHeight is 0.3 / 0.35 * 100 and topSolidLayers is 8 and bottomSolidLayers is 8 in the profile's settings.

I think the 'Support on build plate only' would be similar to enabling 'Use support material'  and 'Dont' support bridges' settings. I not entirely sure though as I'm not very familar with Slic3r's settings. The guy who did all of M33 Fio's Slic3r stuff is @ra100.

The problem with the last used slicer not being remember has been fixed in [M33 Fio's devel branch](https://github.com/donovan6000/M33-Fio/tree/devel). You can install it by with OctoPrint's plugin manager by install a plugin from ths URL `https://github.com/donovan6000/M33-Fio/archive/devel.zip`

NiklasOlofsson, 2017-04-27:
Ah, thanks! I was printing a fan duct and got supports inside the model where the air was supposed to go. With the prusa edititon and support on build plate only fixed the problem.

donovan6000, 2017-04-28:
I looked at the Pursa Slic3r Edition and saw the "Support on build plate only" setting that your talking. You should just be able to add this line, `support_material_buildplate_only = 1`, in the advanced section in the slicer profile editor to enable it.
[image]
That settings not in Slic3r V1.2.9, which is why it wasn't incorporated into M33 Fio's slicer profile editor. However it is in the soon to be released Slic3r V1.3.0, so I'll probably replace M33 Fio's 'Don's support bridges' settings with a 'Support on build plate only' setting after Slic3r V1.3.0 gets released and is supported by [OctoPrint's Slic3r plugin](https://github.com/OctoPrint/OctoPrint-Slic3r). In the mean time, manually adding the setting into the advanced section when you want to enable it is probably the best solution.