donovan6000/M33-Fio #223: Custom control disappear

Fullmono, 2017-05-09:
I don't have access to "custom controls" pannel (calibration, every control made for m3d...) anymore .
I did try to reinstall M33 but it didn't work.

I don't understand why, the controls were present yesterday!

donovan6000, 2017-05-10:
Make sure to disable the Settings > M33 Fio > 'Not using a Micro 3D printer'. M33 Fio V1.18 has that setting enabled by default since it was it was inconveniencing all the M33 Fio users who aren't using Micro 3D printers.

donovan6000, 2017-05-11:
Looks like there's a problem with M33 Fio's JavaScript. Can you post any error messages that you see in your web browser JavaScript console? You can get the the console by right clicking on the page and choosing an options that's similar to 'Inspect Element'. That should open a window that has an option similar to 'Console' that shows if there's any JavaScript errors.
[image]

Fullmono, 2017-05-12:
You are right:

[image]

donovan6000, 2017-05-16:
@Fullmono I think something got messed up with your OctoPrint installation. The gcodeViewModel is one of the view models that OctoPrint has always provided. Try reinstalling OctoPrint and that should fix the issue..

closed 2017-05-16 by Fullmono