donovan6000/M33-Fio #226: Slicer editor

knightofkairell, 2017-06-10:
So the slicer editor is great except for in the scale adjustment Y is Actually changing the Z dimension and and Z dimension is changing the Y dimension.

bobbypop999, 2017-06-11:
I was also woundering if you could add a vase mode to the slicer as that would be very help full.

donovan6000, 2017-06-13:
@knightofkairell I haven't been able to replicate this issue. Scaling a model scales with respect to the model's local axis, and the Z dimension of the model's local axis can identified by the blue cube anchor point. Rotating the model in turn rotates the model's local axis, and doing so can cause the model's local axis to not be aligned with the world's axis. In the image below, I rotate the cube by 90° on its Y axis which caused the cube's local Y and Z axis to be aligned with the world's Z and Y axis respectively. Now scaling the cube in the Z dimensions means the same things as scaling the cube in the world's Y dimensions. Apart from that situation, I can't think of any other way where scaling in the Z dimension could be interpreted as being scaled in the Y dimension.
[image]

@bobbypop999 Vase mode seems like too to specific of a setting to add a checkbox to the slicer editor for, so I probably wont add it in. I tend to accept most of M33 Fio's pull requests though, so feel free to add it in yourself and submit a pull request if you really need in M33 Fio.

Also, you can manually enable vase mode by setting the `spiralize` setting to `True` if your using Cura.
[image]
Or by setting the `spiral_vase` setting to `1` if your using Slic3r.
[image]

donovan6000, 2017-06-14:
Oh, I get it now. There's two common axis syntax, Y-up and Z-up, that get used in the world. I'm more used to thinking in terms of Y-up when working with 3D graphics because I've worked a lot with Autodesk Maya and OpenGL, so that's why M33 Fio's model editor uses a Y-up left-handed coordinate system. It'll all just a matter of preference though.

A good history about the different coordinate systems is explained in [this post](https://forums.unrealengine.com/showthread.php?75420-Z-Up-vs-Y-Up-the-solution-to-the-debate-lies-within&p=326320&viewfull=1#post326320):

> The tale starts with 3DS Max. Released in 1990, it's designed and used primarily for architectural purposes. In Architecture, you typically start with your floor plan which would be plotted on a 2D grid first, X and Y. When it comes to extruding that up into a 3D model, you need the next dimension (Z) to be up. That fellow devs, is apparently why Z is up in 3DS Max. Makes perfect sense for it to be that way.

> In 1998 Maya is released primarily as an animation package. Animation at the time was typically in 2D, from the side-on. Think early Disney and other old animated titles done from a side-on perspective (like a 2D platformer). You start working in X and Y, and your new dimension Z becomes the depth. That would also explain why Maya is Y-Up.

knightofkairell, 2017-06-14:
I see. Well is there a chance of making it possible to change from y-up to z-up? I personally think that z-up is easier to understand for the 3d printer world since everything prints XY and z-up. 

Also I installed the devel branch and now my slicer doesn't display properly. It it missing the next and save buttons they drop off the bottom of the display. This may just be because my screen is wide and short, resolution of the display is 1366x768.

[image]

donovan6000, 2017-06-15:
Ok, the model editor is now Z-up in the devel branch.

I also changed the sizing of the model editor so it should fit your screen again. The addition of the 'Import from server' button made me require a taller screen to make sure that its settings didn't overlap with the printer color buttons. I changed it so that it now requires a wider screen instead, so here's the minimum size of the model editor when using a Micro 3D printer.
[image]
And here's the minimum size when not using a Micro 3D printer.
[image]

knightofkairell, 2017-06-15:
Thanks a lot. Tested and confirmed working.

closed 2017-06-15 by knightofkairell