donovan6000/M33-Fio #227: Skew calibration

knightofkairell, 2017-06-10:
I noticed that the only thing we can use for this is the backlash settings. Backlash is really for correcting motion smoothness. Can we get a skew calibration also? This would be to correct for the gantry being out of square. We would need to set both x and y corrections. Please and thank you. How can I support you in getting this added feature?

donovan6000, 2017-06-13:
[M33 Fio's devel branch](https://github.com/donovan6000/M33-Fio/tree/devel) now implements skew compensation. You can try it out by using OctoPrint's builtin plugin manager to install a plugin from this url: https://github.com/donovan6000/M33-Fio/archive/devel.zip. In Settings > M33 Fio you can set the X and Y skew values.

The skew values should be equal to the total displacement for each axis. For example, the X skew value can be determined by subtracting the X position when the extruder is at the heightest height that it can be from the X position when the extruder is at the lowest height that it can be. In this picture the resulting X skew value is positive.
[image]

M33 Fio's skew compensation pre-processor stage will then compensate for the skew by applying offsets in the opposite direction of the skew values that are proportional to the current extruder's height to all the G-code commands. So if the X skew is positive like in the picture, then the resulting G-code will look like this.
[image]

knightofkairell, 2017-06-13:
That looks like it will fix my issue. But just to be clear. Objects are
being printed out of square. I have attached a picture with a example
dimensions. This is what I am looking to correct. PS sorry my square looks
like a rectangle, it is hand drawn after all.

This results in a parallelogram instead of the square it is supposed to be.
Which also causes circles to become ovals.

Judging from your example you supplied the skew offset for y axis will be
2mm and the skew offset for x axis will be 2mm. I am sure this offset is
larger if you print a larger object. Say a 100mm, that would result in a y
axis skew offset of 4mm and an x axis skew offset of 4mm. Correct?

So how do I determine how much to set the skew offset to?

Thanks for your help.
Alex

donovan6000, 2017-06-14:
The skew offset is related to the max height that the extruder can go, and it's unrelated to the max height of the object begin printed. The equation I'm using to calculate the skew offset is: `extudersCurrentZ / extrudersMaxZ * -skewValue`. That way the skew offset is linearly proportional to the ratio of the extruder's current height and its max height. The `extrudersMaxZ` is 112mm for the Micro 3D printer, and here's what the resulting skew X value is for several heights.

Lowest extruder height: `0 / 112 * -skewX = 0`
Middle extruder height: `56 / 112 * -skewX = -skewX / 2`
Highest extruder height: `112 / 112 * -skewX = -skewX`

Assuming that the cube in your picture is 50mmx50mmx50mm, then you'd want the X skew offset to be -2mm when the extruder is at a height of 50mm. `50 / 112 * -skewX = -2, skewX = 4.48`. The Y skew in your picture is leaning toward the negative Y axis, so you want to make sure it's a negative value: `50 / 112 * -skewY = +2, skewY = -4.48`

knightofkairell, 2017-06-14:
Ok so as far as I can tell this will only make the part print perpendicular in relation to the bed level. That is to say the vertical dimensions will be more square. That is great, but that is not my problem. My x axis doesn't print perpendicular to my y axis. This is why my resulting print of a cube is out of square in both the x and y axis. The z height is not an issue. Z prints perpendicular to both the x and y axis. I am trying say if I start at 0x,0y and move to 0x,50y results in actual position of 2x,50y. This also occurs in x movement. Start at 0x,0y and move to 50x,0y results in actual position of 50x,2y. These movements are supposed to be straight, instead they are diagonal. 

Not sure how to fix this. Is it backlash? I just changed my backlash settings and will test this to see if it has improved at all.

donovan6000, 2017-06-15:
It kind of sounds like your gantry isn't level, like this.
[image]
[Manually leveling your gantry](https://printm3d.com/solutions/article.php?id=210) should help.

knightofkairell, 2017-06-15:
Thanks for that link. It was helpful. I have done this before without the instructions, it was a serious pain. I will take another look at trying this. For now I seem to have it working good enough to get some decent prints. Going to finish the batch I am working on. Thanks for your help. I like how good your plug-in works on OctoPrint. Keep up the good work.

closed 2017-06-15 by knightofkairell