donovan6000/M33-Fio #229: Stuck in bootloader mode?

Fullmono, 2017-06-13:
Hi,

After last update, Octoprint asked to send my bootloader image to developer to help him to reverse engineer the printer. So I did but it failed and now my printer is stuck in bootloader mode with official m3d firmware. What can I do ? 

Edit: I did reboot my printer and now I'm stuck on "detecting baudrate" connection step.

donovan6000, 2017-06-14:
Usually when it's stuck on detecting baudrate it means that M33 Fio crashed so OctoPrint is using its default connection routine. Since your still getting this problem after restarting OctoPrint, then something is persistently causing M33 FIo to crash. Can you upload OctoPrint's log file here so I can see what's causing it? 
[image]

Also just to make sure that there's nothing wrong with your printer, can you to using M33 Manager to connect to your printer, [there's links for each OS in the description of iMe's github repo](https://github.com/donovan6000/iMe), and let me know if it reports any errors. The Linux binary is only for x64 cpus, so if your using a Raspberry Pi then you'll have to recompile it.

Fullmono, 2017-06-15:
Hi, 

- Log is empty
- No problem with official M3D app
- Still "Detecting baudrate" error using Octoprint
- No Problem with M33 Manager. He just says that the printer is in bootloader mode. Is this normal?

donovan6000, 2017-06-15:
Make sure your looking at the octoprint.log file and not the plugin_m33fio_debug.log file. The octoprint.log file shouldn't be empty since it's where OctoPrint logs all its startup and shutdown output to. It should start off the same way as [this one](https://github.com/donovan6000/M33-Fio/files/1076961/octoprint.txt).

It's normal for M33 Manager to say that the printer is in bootloader mode. It doesn't automatically switch the printer into firmware mode like M3D's software or M33 Fio, so it's just saying whatever mode the printer is currently in when it connects to it.

Fullmono, 2017-06-15:
Sorry, by empty I mean nothing relevant.

Here it is:

> 2017-06-14 08:51:46,819 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.1.42
> 2017-06-14 08:52:00,650 - octoprint.plugins.announcements - INFO - Loaded channel _important from http://octoprint.org/feeds/important.xml in 0.28s
> 2017-06-14 08:52:02,636 - octoprint.plugins.announcements - INFO - Loaded channel _releases from http://octoprint.org/feeds/releases.xml in 0.28s
> 2017-06-14 08:52:03,451 - octoprint.plugins.announcements - INFO - Loaded channel _blog from http://octoprint.org/feeds/octoblog.xml in 0.3s
> 2017-06-14 08:52:04,024 - octoprint.plugins.announcements - INFO - Loaded channel _plugins from http://plugins.octoprint.org/feed.xml in 0.095s
> 2017-06-14 08:52:05,048 - octoprint.plugins.announcements - INFO - Loaded channel _octopi from http://octoprint.org/feeds/octopi.xml in 0.26s
> 2017-06-14 08:52:05,740 - octoprint.plugins.announcements - INFO - Loaded channel _m33fio from https://exploitkings.com/scripts/M33 Fio.xml in 0.54s
> 2017-06-14 08:57:26,265 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
> 2017-06-14 08:57:46,298 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
> 2017-06-14 08:57:47,319 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
> 2017-06-14 08:57:48,339 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
> 2017-06-14 08:57:49,366 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
> 2017-06-14 08:57:50,397 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0

donovan6000, 2017-06-15:
I'm not really sure what's causing this then. Do you ever see either of these error messages when you disconnect the printer from your computer while it's stuck on detecting the baudrate? 
[image]

Fullmono, 2017-06-15:
I don't know what I did, but It's now working like a charm.

closed 2017-06-15 by Fullmono