donovan6000/M33-Fio #230: Install failed: Could not parse output from pip

work2fly, 2017-06-14:
I've been running octoprint / octo pi with the M33-Fio plugin for a while now.  It has been working pretty well in between me breaking and fixing the printer (anet a8).

I got notification to upgrade to 1.3.4 today and M33-Fio 1.20 (I think) however the M33-fio plugin failed to upgrade.

I uninstalled the plugin then had a go at re-installing it.

Octoprint pops up a dialog to say:
Installing the plugin from "https://github.com/donovan6000/M33-Fio/archive/master.zip" failed: Could not parse output from pip, see plugin_pluginmanager_console.log for generated output

The plugin_pluginmanager_console.log reports:

2017-06-14 16:06:53,375   /home/pi/oprint/bin/python -m pip install https://github.com/donovan6000/M33-Fio/archive/master.zip
2017-06-14 16:07:09,500 > Collecting https://github.com/donovan6000/M33-Fio/archive/master.zip
2017-06-14 16:07:10,904 > Downloading https://github.com/donovan6000/M33-Fio/archive/master.zip
2017-06-14 16:08:07,435 ! Exception:
2017-06-14 16:08:07,506 ! Traceback (most recent call last):
2017-06-14 16:08:07,510 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/basecommand.py", line 215, in main
2017-06-14 16:08:07,514 ! status = self.run(options, args)
2017-06-14 16:08:07,519 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/commands/install.py", line 324, in run
2017-06-14 16:08:07,525 ! requirement_set.prepare_files(finder)
2017-06-14 16:08:07,535 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/req/req_set.py", line 380, in prepare_files
2017-06-14 16:08:07,538 ! ignore_dependencies=self.ignore_dependencies))
2017-06-14 16:08:07,547 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/req/req_set.py", line 620, in _prepare_file
2017-06-14 16:08:07,549 ! session=self.session, hashes=hashes)
2017-06-14 16:08:07,553 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 821, in unpack_url
2017-06-14 16:08:07,561 ! hashes=hashes
2017-06-14 16:08:07,563 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 659, in unpack_http_url
2017-06-14 16:08:07,565 ! hashes)
2017-06-14 16:08:07,567 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 882, in _download_http_url
2017-06-14 16:08:07,569 ! _download_url(resp, link, content_file, hashes)
2017-06-14 16:08:07,584 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 605, in _download_url
2017-06-14 16:08:07,588 ! consume(downloaded_chunks)
2017-06-14 16:08:07,590 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/utils/__init__.py", line 852, in consume
2017-06-14 16:08:07,602 ! deque(iterator, maxlen=0)
2017-06-14 16:08:07,605 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 571, in written_chunks
2017-06-14 16:08:07,607 ! for chunk in chunks:
2017-06-14 16:08:07,609 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/utils/ui.py", line 139, in iter
2017-06-14 16:08:07,611 ! for x in it:
2017-06-14 16:08:07,624 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/download.py", line 560, in resp_read
2017-06-14 16:08:07,626 ! decode_content=False):
2017-06-14 16:08:07,628 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/requests/packages/urllib3/response.py", line 353, in stream
2017-06-14 16:08:07,633 ! for line in self.read_chunked(amt, decode_content=decode_content):
2017-06-14 16:08:07,642 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/requests/packages/urllib3/response.py", line 502, in read_chunked
2017-06-14 16:08:07,645 ! self._update_chunk_length()
2017-06-14 16:08:07,654 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/cachecontrol/adapter.py", line 106, in _update_chunk_length
2017-06-14 16:08:07,657 ! self._fp._close()
2017-06-14 16:08:07,659 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/cachecontrol/filewrapper.py", line 50, in _close
2017-06-14 16:08:07,673 ! self.__callback(self.__buf.getvalue())
2017-06-14 16:08:07,675 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/cachecontrol/controller.py", line 275, in cache_response
2017-06-14 16:08:07,677 ! self.serializer.dumps(request, response, body=body),
2017-06-14 16:08:07,682 ! File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/_vendor/cachecontrol/serialize.py", line 87, in dumps
2017-06-14 16:08:07,692 ! ).encode("utf8"),
2017-06-14 16:08:07,695 ! MemoryError

I may reset things back to use the default octopi image but before I try that I was wondering if anyone had any suggestions?

cheers
Jamie

donovan6000, 2017-06-14:
The `MemoryError` error indicates that [your device ran out of memory](https://docs.python.org/3/library/exceptions.html#MemoryError). You can try running [pip without caching](https://stackoverflow.com/a/31526029) by adding `--no-cache-dir` to pip's arguments in Settings > Plugin Manager > Settings > Additional Arguments. That might resolve the issue.
[image]

work2fly, 2017-06-14:
it didn't like that option :-(

work2fly, 2017-06-14:
I checked the rasp config settings. the gpu had 128Mb of ram allocated to it. It doesn't really need that (I guess) as I'm not running a display. Swapped this to 32Mb then restarted the pi. The install has now been successful.

thanks for pointing out the obvious memory issue. Not being familiar with the internal workings of octoprint I wasn't sure what or where that error would relate to.

Cheers
Jamie

closed 2017-06-14 by work2fly