donovan6000/M33-Fio #231: Fio causing WebGL to crash

fooforever, 2017-06-21:
Just performing a new install of octoprint on a raspberry pi, using the latest octopi image. After installing the Fio plugin chrome hangs when loading the octoprint page  then webgl crashes. There doesn't seem to be anything printed in any of the debug logs found in the octoprint menus.

Using Chrome 59 on Arch Linux

donovan6000, 2017-06-22:
Are there any JavaScript errors reported in Chrome's console. You can view it by right clicking anywhere, choosing 'Inspect', and then going to the 'Console tab'.
[image]

Also are you experiencing this in any other web browsers, or just Chrome?

fooforever, 2017-06-27:
Sorry @donovan6000 erm it appears to have started to work again.just logged in again to check for errors and its fine.

donovan6000, 2017-06-28:
It's alright. Glad everything's working again.

closed 2017-06-28 by donovan6000