donovan6000/M33-Fio #235: Extracting and uploading firmware failed

dylanmtaylor, 2017-08-08:
I have an original micro (I backed the project), and I tried to upload my bootloader but it just comes back saying this failed.

donovan6000, 2017-08-09:
What operating system are you using, what version of M33 Fio are you using, and what version of OctoPrint are you using?

dylanmtaylor, 2017-08-09:
Raspbian 8.0 Jessie - OctoPi premade distribution on a Pi Zero W. OctoPrint version : 1.3.4
OctoPi version    : 0.14.0 M33 Fio version : 1.21

donovan6000, 2017-08-12:
Can you upload your OctoPrint log so we can see if it contains any errors related to M33 Fio? You can get it it at Settings > Log > octoprint.log.
[image]

closed 2017-10-18 by donovan6000