donovan6000/M33-Fio #236: Support for M3D Pro?

TheNetStriker, 2017-08-14:
Does this plugin also support the new M3D Pro?

donovan6000, 2017-08-14:
No, M33 Fio doesn't support Micro+ or M3D Pro printers.

TheNetStriker, 2017-08-14:
Thanks for the quick answer. Is it planed to support those printers in the future?

donovan6000, 2017-08-14:
I don't plan on buying either of those printers, so I don't have much  of a reason to add support for them. So if support is added for those printers, it probably wont be done by me.

xcasex, 2017-09-27:
what if someone were to send you a micro+ ? :)

sci-phi, 2017-10-04:
Any pointers on getting started with M33 Fio if someone else wanted to add support for Plus/Pro?

Does it require reverse engineering something new, or is it more like avoiding a breaking change/bug and updating for new features?

bas-velthuizen, 2017-10-20:
I'm with sci-phi. If you can give any pointers where to start for adding support for the Micro+ (I've gooten mine this week) I'd be glad to help.

donovan6000, 2017-10-21:
Adding support for the Micro+ or M3D Pro printers requires reverse engineering M3D's software and those printers' firmware and bootloader. There's a lot that we don't know about those printers since there hasn't been much interest from the community in reverse engineering them, so getting to the point where we know as much about them as we do the Micro 3D printer is a very ambitious project.

The main thing needed to get OctoPrint to successfully connect to them is to alter [M33 Fio's connection routine](https://github.com/donovan6000/M33-Fio/blob/master/octoprint_m33fio/__init__.py#L7261) so that it works with those printers. Those printer's bootloaders behave slightly different that the Micro 3D's, so it's crucial that you know how to use the bootloader to manage the printer's EEPROM and firmware.

Knowing what all the offsets in the Micro+ and M3D Pro's EEPROM mean is also important since they differ quite a bit from the [Micro 3D's EEPROM](https://github.com/donovan6000/M33-Fio/blob/master/octoprint_m33fio/__init__.py#L171). All of the code in M33 Fio that refers to the EEPROM values will have to be adjusted to account for differences depending on the printer being used. The same is true for all of the [Micro 3D specific settings](https://github.com/donovan6000/M33-Fio/blob/master/octoprint_m33fio/__init__.py#L2192). The Micro+ and M3D Pro will have some settings that the Micro 3D doesn't and will lack some that the Micro 3D has, so the code will also have to be changed to account for this.

Another thing is that some of the [G-code pre-processor stages](https://github.com/donovan6000/M33-Fio/blob/master/octoprint_m33fio/__init__.py#L10377) will need to disabled depending on if the user is using a Micro+ or M3D Pro since those printer's have some builtin features, like builtin bed compensation, that the Micro 3D doesn't.

That covers most of the backend stuff. After that there's a lot of aesthetic stuff, like Micro 3D specific options that become visible when the user is using s Micro 3D printer, that will need to also become visible/invisible depending on if they are applicable the the Micro+ or M3D Pro. Most of that stuff is handled in [this file](https://github.com/donovan6000/M33-Fio/blob/master/octoprint_m33fio/static/js/m33fio.js).

LeonFedotov, 2019-07-06:
how about replacing the internals with something compatible?