donovan6000/M33-Fio #237: Z-axis pops off threads after large print

A-Smith, 2017-08-27:
I don't know if this is coming from M33-Fio, OctoPrint itself, or Cura, but I thought I'd start by reporting it here: After a print, the print head moves up, out of the way of the model. Normally, this is a good feature, but if the model is already near the max printable height, this can cause the rack to pop off the ends of the z-axis driving rods. This has happened to me twice so far. All of my OctoPrint and M33-Fio settings are at the default values, except that I have cleared the "Not using a Micro 3D printer" checkbox, so the software should be aware of the valid M3D print volume. I checked the GCode generated by Cura for the last model I printed: The last printed layer is at Z78.150, and the end of the .gco file is:
```
;LAYER:520
M107
G1 F1200 E700.92528
G1 Z78.250
G0 F1080 X52.021 Y57.861 Z83.333
```
That should be well within the safe print volume. That leads me to believe that "moving the print head out of the way" is done by something other than the slicer. I didn't see an OctoPrint or M33Fio setting that would let me disable that. Can you either point me in the right direction, or take this as a feature request to have M33Fio cause OctoPrint to "do something safe" at the end of the print, when the model is very tall?

I'm uisng OctoPrint 1.3.4, M33-Fio 1.21, and Cura 15.04.6. I'm running on 64-bit Windows 10.

donovan6000, 2017-08-30:
This is a result of M33 Fio's preparation preprocessor. [It moves the extruder to at most 10mm above the extruder's highest position in the print when the print is done](https://github.com/donovan6000/M33-Fio/blob/master/shared%20library%20source/preprocessor.cpp#L1856-L1859) which should put yours at `Z83.333 + 10 = Z93.333`. 93.333mm is well within the the max advertised Z of 116mm, so I'm not sure why your experiencing this.

If your using iMe firmware, then make sure that the `Z motor steps/mm` setting is set to `646.3295 steps/mm`. If that's not causing the problem then you can increase the `External bed height` setting to something like `20mm` which would make is so that the extruder wouldn't be able to go higher than 92mm while printing. `112mm - 20mm = 92mm`.

A-Smith, 2017-08-31:
Sorry, I should have mentioned the firmware version. I'm using stock M3D firmware, v2016040401. What you say makes sense. From the code you linked to, I see M33 Fio should already be limiting it to a safe value. At the end of the print, it looked to me like the print head lifted more than 10mm, though. I don't know why. Maybe some other part of OctoPrint is also trying to raise it... 

Regardless, the next time I try to print something tall, I'll try setting a "fake" external bed height to limit the Z height. Thanks for the suggestion. Incidentally, thanks for making M33 Fio! I much prefer the combination of OctoPrint/Cura/M33 Fio to M3D's own software.

closed 2017-08-31 by A-Smith