donovan6000/M33-Fio #241: Memory Error

rohanantao, 2017-11-26:
I have installed Octoprint 1.3.5 and trying to install M33 Fio through the plugin manager.

I keep getting a memory error and then the system stops.

--- plugin log ---
2017-11-26 18:21:36,720   /usr/bin/python2.7 -m pip install https://github.com/donovan6000/M33-Fio/archive/master.zip
2017-11-26 18:21:49,194 > Collecting https://github.com/donovan6000/M33-Fio/archive/master.zip
2017-11-26 18:21:59,685 > Downloading https://github.com/donovan6000/M33-Fio/archive/master.zip
2017-11-26 18:22:20,426 ! Exception:
2017-11-26 18:22:20,437 ! Traceback (most recent call last):
2017-11-26 18:22:20,439 ! File "/tmp/pip-build-37p9J5/pip/pip/basecommand.py", line 215, in main
2017-11-26 18:22:20,441 ! File "/tmp/pip-build-37p9J5/pip/pip/commands/install.py", line 324, in run
2017-11-26 18:22:20,449 ! File "/tmp/pip-build-37p9J5/pip/pip/req/req_set.py", line 380, in prepare_files
2017-11-26 18:22:20,451 ! File "/tmp/pip-build-37p9J5/pip/pip/req/req_set.py", line 620, in _prepare_file
2017-11-26 18:22:20,463 ! File "/tmp/pip-build-37p9J5/pip/pip/download.py", line 821, in unpack_url
2017-11-26 18:22:20,466 ! File "/tmp/pip-build-37p9J5/pip/pip/download.py", line 659, in unpack_http_url
2017-11-26 18:22:20,468 ! File "/tmp/pip-build-37p9J5/pip/pip/download.py", line 882, in _download_http_url
2017-11-26 18:22:20,470 ! File "/tmp/pip-build-37p9J5/pip/pip/download.py", line 605, in _download_url
2017-11-26 18:22:20,483 ! File "/tmp/pip-build-37p9J5/pip/pip/utils/__init__.py", line 852, in consume
2017-11-26 18:22:20,485 ! File "/tmp/pip-build-37p9J5/pip/pip/download.py", line 571, in written_chunks
2017-11-26 18:22:20,488 ! File "/tmp/pip-build-37p9J5/pip/pip/utils/ui.py", line 139, in iter
2017-11-26 18:22:20,490 ! File "/tmp/pip-build-37p9J5/pip/pip/download.py", line 560, in resp_read
2017-11-26 18:22:20,492 ! File "/tmp/pip-build-37p9J5/pip/pip/_vendor/requests/packages/urllib3/response.py", line 353, in stream
2017-11-26 18:22:20,500 ! File "/tmp/pip-build-37p9J5/pip/pip/_vendor/requests/packages/urllib3/response.py", line 502, in read_chunked
2017-11-26 18:22:20,502 ! File "/tmp/pip-build-37p9J5/pip/pip/_vendor/cachecontrol/adapter.py", line 106, in _update_chunk_length
2017-11-26 18:22:20,515 ! File "/tmp/pip-build-37p9J5/pip/pip/_vendor/cachecontrol/filewrapper.py", line 50, in _close
2017-11-26 18:22:20,517 ! MemoryError

--- end plugin log ---

Any help would be amazing.

donovan6000, 2017-11-27:
Based on [this answer](https://stackoverflow.com/a/31526029), you should be able to install the plugin if you set pip to not cache the file by using the `--no-cache-dir` flag.

So you set this flag in OctoPrint's settings > Plugin Manager > Settings > Additional arguments
[image]

Or manually run this command to install the plugin.
`/usr/bin/python2.7 -m pip --no-cache-dir install https://github.com/donovan6000/M33-Fio/archive/master.zip`

rohanantao, 2017-11-28:
@donovan6000 - thanks so much for that. I had to run it via the command line as it kept stalling via the Plugins installer. However, now I can connect to the printer, but it connects only when 'Not using a Micro 3D printer' is enabled. Cycled the printers power and tried multiple times (as well as system reboots), but it didnt work.

[--- when Not using m3d printer is disabled --]
Changing monitoring state from 'Operational' to 'Offline'
Connection closed, closing down monitor
Connecting to: /dev/ttyACM0
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0x228d3c0, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=None, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: N0 M110 N0*125[-- END when not using m3d printer is disabled--]

[--- when Not using m3d printer is enabled --]
Connecting to: /dev/ttyACM0
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0x2103e70, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: N0 M110 N0*125
Recv: ok 0
Changing monitoring state from 'Connecting' to 'Operational'
Send: N0 M110 N0*125
Recv: ok 0
Send: N1 M115*39
Recv: ok 1
Send: N2 M21*18
Recv: PROTOCOL:RepRap FIRMWARE_NAME:iMe FIRMWARE_VERSION:00.00.01.25 MACHINE_TYPE:Micro_3D EXTRUDER_COUNT:1 SERIAL_NUMBER:BL16061002025024
Recv: ok 2
Send: M115
Recv: ok
Recv: PROTOCOL:RepRap FIRMWARE_NAME:iMe FIRMWARE_VERSION:00.00.01.25 MACHINE_TYPE:Micro_3D EXTRUDER_COUNT:1 SERIAL_NUMBER:BL16061002025024

[--- when Not using m3d printer is enabled --]

Any suggestions?