donovan6000/M33-Fio #242: Uploading bootloader failed

npag, 2017-11-26:
Just updated my iMe firmware on the Micro and M33-Fio asked me to upload the bootloader of my printer to help with development. I klicked yes but the upload failed. Any way to upload it manually?

donovan6000, 2017-11-27:
If you kill the OctoPrint process, edit the `~/.octoprint/config.yaml` file to change the `Micro3DBootloaderVersionsUploaded` setting to `''`, and then restart OctoPrint then you'll eventually be asked again if you want to upload your bootloader.

But it's not that important, so you shouldn't worry too much about making sure that I get your printer's bootloader. I mainly look at the bootloaders to see if M3D changed anything about the Micro 3D printer, which they haven't done in a while. I appreciate you wanting to help though :)

npag, 2017-11-27:
Okay thank you for the information! Consider this issue closed. Good luck in the future, I really like your project.

closed 2017-11-27 by npag