donovan6000/M33-Fio #253: Chaning cura edge starting postion

guysoft, 2018-10-15:
Hey,
So when Cura on M33-Fio starts it goes all the way to the front left of the printer.
However, in order for my M3D to work I had to add a fan to cool the gears, I used this part:
https://www.thingiverse.com/thing:1704142

Now the fan hits the front left of the build frame. So I want to limit the extruder travelling there.

I tried setting the bed size to 70mm on x and y as a test, re-slice, and try again, but it still  goes to that position.

Any way to enforce this limitation?

donovan6000, 2018-10-18:
M33 Fio doesn't provide the ability to change the bed size when using an Micro 3D printer. You can disable the 'Use preparation preprocessor' setting to prevent M33 Fio from injecting the front left movement at the beginning of a print. However this will also prevent it from injecting a fan speed command at the beginning of a print, so make sure that your g-code has a fan speed command already in it and that you disable M33 Fio's 'Remove fan commands' settings.

If you don't want to disable M33 Fio's preparation preprocessor, then as a workaround you can modify [this line](https://github.com/donovan6000/M33-Fio/blob/master/octoprint_m33fio/__init__.py#L10708) to be 'true' if your not using the shared library, or [this line](https://github.com/donovan6000/M33-Fio/blob/master/shared%20library%20source/preprocessor.cpp#L1789) to be 'true' if you are using the shared library. Modifying the shared library will require you to recompile it. This change will cause the the front left movement to be replaced with M33 Fio prepping the nozzle by extruding 0.4mm of filament in the center of the bed.

guysoft, 2018-11-18:
Ok, in the end my workaround was to just remove the enclosures, that reduced the temperature enough for the printer to print correctly.

I never got this to work, I tried two workarounds, and both gave strange results that didn't seem to do what I wanted. I think these commands were also used in calibration or something like that, and disturbed that too.

closed 2018-11-18 by guysoft