donovan6000/M33-Fio #255: Slicer profiles and editor don't work/show on OctoPrint 1.3.12

jmichiel, 2019-11-13:
For use with my M3D Micro, I recently installed OctoPi 0.16.0 on a RPi 2, which is still using OctoPrint 1.3.10.
After first-time configuration and restarting, it immediately informs you that there is a new version of OctoPrint (1.3.12), so I updated. I then installed M33 Fio and had to install CuraEngine Legacy since CuraEngine has apparently been unbundled from OctoPrint.
However, I didn't have the M3D slicer profiles and the slicer profile editor only shows the Advanced Settings (density and fill settings not showing). 
I restarted from scratch without doing the update (didn't need CuraEngine Legacy, obviously) and then everything seems to look as in the screenshots.

Is it because CuraEngine has been renamed to 'CuraEngine Legacy' that M33 Fio gets confused about what slicer is being used? I'll refrain from updating for the time being...

Haven't had the time to actually print anything, but really looking forward to trying out something else than the standard M3D software ;-)

donovan6000, 2019-11-14:
The last version of OctoPrint that I fully tested M33 Fio in was version [1.3.5](https://github.com/foosel/OctoPrint/releases/tag/1.3.5). I doubt that it will fully work in any version of OctoPrint newer than that.

jmichiel, 2019-11-19:
Ah, that explains a lot ;-)
I'll try with OctoPi 0.14.0 which uses OctoPrint 1.3.4.

sommo, 2020-01-02:
Hi, same problem here. have you found the link of OctoPi 0.14.0 which uses OctoPrint 1.3.4.
Tried to downgrade the octopi manualy from the lastest version but it messed up things..

IonicEV, 2020-01-04:
> The last version of OctoPrint that I fully tested M33 Fio in was version [1.3.5](https://github.com/foosel/OctoPrint/releases/tag/1.3.5). I doubt that it will fully work in any version of OctoPrint newer than that.

I was a bit puzzled when I saw no slicer in latest OctoPi/Print version. Good that I look at github repo.

OctoPi Version 0.15.1 is working fine for me with the cura engine slicer (OctoPrint 1.3.8).  
@donovan6000 I could not find a words to thank you for your work on M3D micro and M33-Fio plugin! Your work alone put the whole M3D dev team to shame. Great calibration and embedded help with images of test print, makes it easy to interpret the results. iMe firmware is another marvel - it transforms dysfunctional M3D micro to a great 3D printer for small jobs.

donovan6000, 2020-01-05:
@IonicEV Thanks! It wasn't just me involved though. Three years ago when open source movement for the M3D printer was in full swing there was a whole community helping test and develop ~~M3D~~ M33 Fio and iMe. That community was the real driving force behind all of it, and it's sad to see how none of that information can be accessed anymore. It's too bad that M3D got rid of their [forum](https://printm3d.com/community/) a while ago and have now blocked off their Discord channel. There was years of documented progress on there that they've destroyed.

Like who gonna remember when [Loswave got the Micro 3D printer working with Repetier-Host](https://www.youtube.com/watch?v=EzlbZfqQTnM) and pretty much kick started the whole Micro 3D open source movement? Or when tomasf was the first person to get the Micro 3D printer working on macOS which eventually resulted in him developing [MicroPrint](http://tomasf.se/projects/microprint/)? Or when M3D's head of customer service, Robert Marietta, demanded that we change the name of our software despite [their team publicly acknowledging M3D Fio in one of their Kickstarter updates](https://www.kickstarter.com/projects/m3d/the-micro-the-first-truly-consumer-3d-printer/posts/1574463). Or when someone decided to trace out the connection on the Micro 3D's motherboard and share it with the community. I based so much of iMe's code off of this one image that was shared in that forum.
[image]

Oh well. It was a great time for those involved, and it's tragic how M3D seems to have now intentionally made it impossible for a community like that to exist around their products. It's definitely ironic how M3D now embraces open source software in all their newest products when they've made several attempts in the past to hinder the progress of open source software for their products. The world definitely realized that M3D's software was mediocre several years before M3D ever did. Sorry to rant for a bit. Your post made me feel a bit nostalgic. Thank you.

IonicEV, 2020-01-05:
FYI, 
Although OctoPi Version 0.15.1 appears to be working, but M33-Fio plugin have some issue with auto-connect (error when reconnecting after reboot, when printer is already in firmware mode) and job pause/cancellation. I rolled back to OctoPi Version 0.14.0 and it working fine now.

sommo, 2020-01-07:
Hi, can't find the iso for OctoPi Version 0.14.0 that work for rpi, could you link me the image file?
Thank you

sommo, 2020-01-07:
i think the problem is that i have a rpi3b+..

sommo, 2020-01-11:
Yes it was the rpi I mounted the image into a rpi2 and all its ok.. cant use on a rpi3

sommo, 2020-01-11:
You can close the case