donovan6000/M33-Fio #256: Issue with Octoprint 1.4.0 or newer - print not starting

Thomas-74, 2020-03-11:
Hello,

I'm running a M3D Micro using the M33 Fio plugin. When I start a print, the printer doesn't move and doesn't heat up. In the octoprint support forum, I was told that this is related to a keyword not being supported anymore by Octoprint 1.4.0. or newer:
[https://community.octoprint.org/t/print-not-starting/16950/3](url)

Here's a log from octoprint:

2020-03-08 20:49:17,727 - octoprint.util.comm - INFO - Changing monitoring state from "Operational" to "Starting"
2020-03-08 20:49:17,750 - octoprint.printer.standard.job - INFO - Print job started - origin: local, path: Kappe.gco, owner: Thomas Heuer, user: Thomas Heuer
2020-03-08 20:49:17,776 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2020-03-08 20:49:17,816 - octoprint.plugin - ERROR - Error while calling plugin m33fio
Traceback (most recent call last):
  File "c:\octoprint\venv\lib\site-packages\octoprint\plugin\__init__.py", line 224, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "c:\octoprint\venv\lib\site-packages\octoprint_m33fio\__init__.py", line 6903, in on_event
    if payload["filename"] == os.path.basename(self._file_manager.path_on_disk(octoprint.filemanager.destinations.FileDestinations.LOCAL, "test border.gcode").replace("\\", "/")) :
KeyError: 'filename'
2020-03-08 20:49:18,006 - octoprint.plugins.tracking - INFO - Sent tracking event print_started, payload: {'origin': u'local', 'file': 'd72586cf622fae1ae9bb65a28b39e2ca5e518e6c'}
2020-03-08 21:00:21,131 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2020-03-08 21:00:30,111 - octoprint.plugins.tracking - INFO - Sent tracking event ping, payload: {'octoprint_uptime': 909}

And here's what's happening in the M33 Fio plugin:

2020-03-08 20:49:17,477 Processed Response: 
2020-03-08 20:49:17,809 Original Sent: N0 M110 N0*125

2020-03-08 20:49:18,479 Original Response: wait

2020-03-08 20:49:18,480 Processed Response: 
2020-03-08 20:49:18,482 Original Response: ok  PT:98 DT:1

2020-03-08 20:49:18,483 Processed Response: ok  PT:98 DT:1

2020-03-08 20:49:20,477 Original Response: wait

2020-03-08 20:49:20,477 Processed Response: wait

2020-03-08 20:49:22,476 Original Response: wait

2020-03-08 20:49:22,476 Processed Response: 
2020-03-08 20:49:22,487 Original Response: ok  PT:98 DT:1

2020-03-08 20:49:22,487 Processed Response: ok  PT:98 DT:1

2020-03-08 20:49:23,473 Original Response: wait

2020-03-08 20:49:23,474 Processed Response: wait

That continues until I abort Octoprint with Strg+C.

donovan6000, 2020-03-12:
The last version of OctoPrint that I fully tested M33 Fio in was version 1.3.5. I doubt that it will fully work in any version of OctoPrint newer than that.

Thomas-74, 2020-03-13:
Thank you for that information. Do you plan an update to fix that compatibility issue or is the development stopped?

donovan6000, 2020-03-14:
I don't own a Micro 3D printer anymore, so I doubt I'll ever make M33 Fio compatible with any newer versions of OctoPrint.

Thomas-74, 2020-03-14:
OK, thanks for telling. So I shall stick to the older version of Octoprint, then.

closed 2020-03-14 by Thomas-74